NiryoRobotics/niryo_one #1: List of not print parts

photo-mickey, 2017-12-28:
Hello!

Can you share the list of not print parts? 
Can I buy your custom electronics parts? I mean adaptor for Raspberry pi and Arduino controllers for steppers. Or maybe you can share the PCB and list of components.

renarded, 2018-01-05:
Hi,

We plan to sell the custom electronics parts from March, 2018. 

We'll also update this repository with more info about schematics, list of parts, assembly guide, tutorials, etc :)

photo-mickey, 2018-01-09:
Thank you. I will wait for it ;) If i have understood correctly, I should wait for new stl too.

Jean8marc, 2018-02-06:
Bonjour,
J'espère que la mise a jour des fichiers imprimables va bientôt arriver. Il me tarde de commencer.

renarded, 2018-02-06:
Hi,

All STL files have been updated (+ tools), and we also add the complete list of materials in the BOM folder. (an assembly video tutorial will be published soon on our website)

You can now print the V 1.0 Niryo One :)

closed 2018-02-06 by renarded