NiryoRobotics/niryo_one #2: previous version stl files

cool45, 2018-01-22:
Hello niryo robotics team I would like to ask where I can find the 3d models for the older servo design as I already do have a lot of servos from previous robotics projects .Thank You for your time :)
,kya lynch﻿

renarded, 2018-01-29:
Hi,

Sorry but this version is not available (and I don't think we can even find it, because we changed it a lot and started to keep strong version archives not so long ago). We will keep updating the current version and published future versions, but not past ones.

The older servo design was dropped for good reasons to switch to Dynamixel XL-320. ([Here](https://www.kickstarter.com/projects/niryo/niryo-one-an-open-source-6-axis-robotic-arm-just-f/posts/1859166) is the update news)

In the meantime we have modified all the other parts of the robot a lot, and the version we've just published is the first stable version. So if you have the version from one year ago, you won't be able to do anything with it. The code on the electronics boards won't work, the dimension of the robot will be wrong, and you'll face heat issues on the motors.

I hope you understand !

closed 2018-01-29 by renarded

jerinpeter, 2019-07-10:
sir is there any way we can get the stl file of fore arm top with the compatibility of normal servos, please do help sir i have spend a loot of time 3d printing the parts and found this out at last thinking that it was backward compatible. can you please please help me... i have been working on this for a month now, as i am building from scratch, gathering parts was also a tiresome job. Hope you will help me.

jerin

renarded, 2019-07-11:
Hi,

Unfortunately the answer I previously provided earlier for this discussion stays the same.

The only possibility for you is to use the STEP files and modify them + print 3D parts, so you can build an alternative robot for your own use.

pawaz123, 2024-09-23:
ok sir can you attach the link of step file