NiryoRobotics/niryo_one #3: BOM

photo-mickey, 2018-02-07:
Hello!

I have some issues about BOM:

1. What is the voltage and size of fan
2. Which switch do you use?
3. What is the size of aluminium tube?
4. Which is diameters of coupler?
5. Which is diametor and teeth nomber of  gear GT2
6. Which is length of belts?
7. Which is power of power supply?

Jean8marc, 2018-02-07:
Bonjour,
Tout d'abord , très bon travail : félicitations.
Par contre, je voudrais ajouter a cette liste ci dessus  d'autres points 
Ribbon ? combien de pins ?
Wiring 1 & 2 : ?
le premier switch ?
DC connector + cable ?
Les "Niryo" moteurs ? est il possible d'avoir des infos ?

J'ai commencé les impressions , et déjà commander quelques pièces.

photo-mickey, 2018-02-08:
Jean8marc, they wrote, that custom electronik will be sold in March. If I understood right, the steppers has oun PCB with encoder, driver, MCU, fan and CAN transmiter. The red stepper has planet gear box (from pic). 
Ribbon - IDC16 (1 to 1) from pic
DC connector like [this](https://ru.aliexpress.com/item/5PCS-LOT-DC-022B-DC-Power-Jack-Socket-dc-connector-Adapter-interface-DC022B-5-5X2-1MM/32800312541.html?ws_ab_test=searchweb0_0,searchweb201602_4_10152_10151_10065_10344_10068_10342_10343_10340_10341_10543_10084_10083_10618_10307_10301_5722313_10313_10059_10534_100031_10103_10624_520_10623_10622_10621_10620_10142_10125,searchweb201603_1,ppcSwitch_5&algo_expid=82093a1f-df77-4c6b-bbb6-6b943442e939-14&algo_pvid=82093a1f-df77-4c6b-bbb6-6b943442e939&transAbTest=ae803_5&priceBeautifyAB=0). You will choose connector after coosing power supply.

renarded, 2018-02-13:
Hi,

We have taken all your questions, and will try to answer them in the next tutorial/video we will publish over the last few weeks. I will close this issue once the tutorials are published.

The Niryo Steppers contains a stepper motor (NEMA) with a magnet on the back, a custom shield where we use AS5600 sensor to get the position, an ATSAMD21G18 microcontroller (same as on Arduino M0), and a MCP2515 for SPI to CAN connection.

photo-mickey, 2018-02-13:
Ok, thank you!
Unfortunately, answers to these questions are wery important for purchase of components.

Why are use SPI to CAN (not USART to CAN)? It isn't critics.

Agent-Cain, 2018-03-15:
Is there a date for the update of the BOM ?
Can I already begin to print the things or will there be major changes in the .stl files ?

AJakob1983, 2018-04-02:
What about the length of the belt loop and aluminium tube?

Do you have some specifications for the torsion spring?

renarded, 2018-04-03:
Hi everyone, 

Sorry for the late reply. We've just published an updated BOM on a Google Sheet, and provided the [link](https://docs.google.com/spreadsheets/d/e/2PACX-1vS6EmT4n43JWafJPHNZq2qyIoZx4T4H3hvxoW7IlD-_LJswFsSpjLghzwzgzjj_bPZjsMo5IDen6eoL/pubhtml?gid=514682279&single=true) in this repo. All the questions asked here have been taken into account :)

@photo-mickey We used SPI to CAN because we were already using the UART pins of the Raspberry Pi 3 for the Dynamixel Bus communication (see schematics [here](http://support.robotis.com/en/product/actuator/dynamixel_x/xl-series_main.htm))

closed 2018-04-03 by renarded

photo-mickey, 2018-04-05:
Hello!
1. There is no pic in doc
2. "Gear GT2 : 16 teeths" - diametor is 5mm?
3. "Half Thurst Bearing" - what is it?
4. Do you have any wire drawings?