NiryoRobotics/niryo_one #4: stl assemblies

smaassen, 2018-02-07:
Hi,

Is it possible to get the `.stl` assemblies of the different part and maybe one assembly of the entire robot? My specific application needs the dimensions of the gripper assembly in order to get the transformation from hand to gripping point. It is quite tedious to near impossible to assemble the `.stl` parts in a cad software (to many faces loaded at once) and measuring it by hand is very uncertain.

I think this will also be useful for the community if they want to design custom parts (custom grippers, sensor mounts, other modifications) Even better would be to publish `.step` files for easier import in CAD software, if you want of go the route of open source. This would be still be an abstraction layer to say Solidworks parts, since all history action is lost

Best

renarded, 2018-02-13:
Hi, 

We are currently working on that, this will be published on this repository soon, so you can fully customize your robot.

renarded, 2018-09-11:
Hi @smaassen,

We have just published the .step files and more useful assembly files under the ["integration" folder](https://github.com/NiryoRobotics/niryo_one/tree/master/Integration).

closed 2018-09-11 by renarded