NiryoRobotics/niryo_one #5: Electronic layout

RauLasH26, 2018-02-27:
Hi,

Is it possible to get the PCB layout for "Niryo shield" and the pannel with all the parts that I need to build one to myself. It will be nice if you put an electronics quide (which cable goes where)  for this robot. If you can't for this new upgrade version, please do it for the old version the one with the Arduino Mega. Thank you so much, wish you all the best guys!

renarded, 2018-02-27:
Hi, 

We'll provide you with basic schematics of the RPi Niryo shield (also the Niryo Stepper boards), so you can see what is connected to what (with RPi GPIOs), and how to use the connections with different devices/cables. This doc is in our TO-DO list, we'll publish it soon. However, we won't release the full PCB file for those boards.

The cables for the old version with Arduino mega is very simple. We used a RAMPS 1.4 shield, with standard stepper drivers, and directly connected the steppers to the shield. Back then we used hobby servomotors that were connected at the place reserved for servos. (we later changed to Dynamixel and new shield almost at the same time, and Dynamixel motors directly on the RAMPS 1.4 was not really a success). Then the communication between the Arduino Mega and Raspberry Pi was I2C, which was not much stable then, you can also try with UART. That's pretty much everything you need to know about the old architecture. But note that with this architecture, you won't be able to use the current code (ROS + Arduino) that we wrote. You will need to make a lot of modifications.

RauLasH26, 2018-03-06:
Thank you for all the information. I have another request, can I buy only those 2 boards. How much will it cost me?

renarded, 2018-03-06:
You'll be able to buy those 2 boards (RPi Niryo Shield and Niryo Stepper board) after we officially release Niryo One, in about 2 months. We'll open a new store on our website so you can get every separate parts of the robot.

Stay tuned !

closed 2018-04-03 by renarded