NiryoRobotics/niryo_one #8: Shield et PCB

Furansujin, 2018-10-13:
Bonjour,

Quand sera t'il possible d'acheter vos custom Shield et PCB ? 

Merci Beaucoup et bonne continuation dans votre projet.

Furansujin, 2018-10-13:
ou ajouter les plans pour le fabriquer ? car sa fait longtemps qu'on attend .
sniff

renarded, 2018-10-16:
Bonjour @Furansujin,

Nous n'allons pas rendre publique les plans de fabrication des shields et PCB. Vous pouvez quand même voir la schématique globale dans le dossier "Electronics" de ce repo.

Concernant l'achat des custom Shield et PCB, nous ne proposons pas pour l'instant de les acheter à l'unité (problèmes d'organisation de production et de stock autrement). Vous pouvez toutefois acheter un Maker Kit sur notre site et vous aurez ces cartes dans le kit.

closed 2018-10-16 by renarded