NiryoRobotics/niryo_one #13: raspberry image

marcostrullato, 2019-09-21:
Hi, I'd have a couple of questions about the software components on the raspberry.
- where can I find documentation about what's running on the raspberry image?
- do you install some proprietary software or is it just ROS?

Thanks

marcostrullato, 2019-09-21:
Ah, by the way, I have downloaded the latest image available and put into an SD as usual.
Unfortunately it doesn't boot, the screen hangs on the rainbow image.
My hardware is a raspi3.

Do you have suggestions?

Thanks

renarded, 2019-09-23:
I guess you're using a Raspberry Pi 3B+.

As for now the image is only compatible with Raspberry Pi 3B. We'll put some efforts in the future to make it compatible with version 3B+ and/or 4 (can't tell the date exactly).

As for documentation, you can find all the ROS packages [here](https://github.com/NiryoRobotics/niryo_one_ros) :)

closed 2019-09-23 by renarded