NiryoRobotics/niryo_one #16: Trouble with Niryo Shield and Niryo Panel

WilliamH07, 2020-04-11:
Hello,
I am a young robotics developer and today I started building Niryo One but here I am missing 2 essential components for the robot it’s 2 components are the Niryo Shield and the Niryo Panel but unfortunately I don’t found nowhere on the internet or on Niryo’s site
Can you help me ?
Thank you
Regards WilliamH07

MikeMihailov, 2020-04-11:
Hello, WilliamH07!

There are a lot of components, that you can't find here. You can buy it from Niryo or join the community of makers: https://forum.movimentomaker.pt/t/braco-robotico-3d-niryo-one/1060/183 and https://forum.movimentomaker.pt/t/niryo-modificacoes/1127/98

WilliamH07, 2020-04-11:
Tank you @MikeMihailov

closed 2020-04-11 by WilliamH07